RealTadango/FrSky #1: feature request for ECU

geofrancis, 2017-08-09:
can you get the throttle and engine status from the Xicoy ecy?

RealTadango, 2017-08-16:
Engine status can be added, but is must be translated in the TX from a status byte to a text value. A lua mixer script can do that. I will add it later (should be something possible for Jetronic also i think (maybe make a universal status list?)

But why do you want throttle status? FrSky is logging Throttle input from the TX already?

geofrancis, 2017-08-16:
So it can show if the throttle is at engine start position etc

alex998r, 2017-08-17:
Hello I am studying the Jetcat ecu serial protocol to interface the Frsky telemetry system, do you have some details about the protocol? Or maybe you already have done this.
Thanks

RealTadango, 2017-08-18:
@alex998r I don't have a jetcat ecu available and i don't know the protocol. If you have any info available to share, please do :) Maybe open another issue for jetcat compatibility.

alex998r, 2017-08-18:
Hello, I have a document that explains some cmd/reply from the ECU.

I do have a Jetcat p120sx turbine with ECU V6, so I can play with it, I have to search a suitable ttl/rs232 converter in my laboratory to sniff the frames coming out from the ecu.

It is essentially an rs232 protocol ( 5V ttl ). The bus has two interfaces, I2C to communicate with the slaves and RS232 for their program.

We can convert the replies from the ecu to SBUS frsky telemetry.

Please find attached the document.

Ciao from Italy

Alessandro

alex998r, 2017-08-18:
Hello, here some informations that can be useful:

https://github.com/raphaelcoeffic/sport_ecu

timercop, 2022-12-13:
How to collect engine status from the Xicoy ecu? 
I've seen some commented out lines in the Ecu_Fadec.cpp file that appear to be status lines but I couldn't detect how to do this collection.