RealTadango/FrSky #2: GPS 2 Sensor from Power Box to implement

hanskranzler, 2018-03-23:
Hello Real Tadango
Would it be possible to change the program so that only the GPS2 sensor works.
It is the best GPS sensor for a jet.
It is also a log how far and how high was flown, a problem on our airfield.
[image]
Best regards
Hans Kranzler

RealTadango, 2018-03-25:
Yes i will create a sensor for that. I already ordered a PB GPS 2 for that. The FrSky GPS sensor is fun but far from accurate.....

hanskranzler, 2018-03-26:
Great.

hanskranzler, 2018-04-09:
Hello
How is the status?

RealTadango, 2018-04-09:
I received the PB GPS 2 yesterday and will be building my new development machine tomorrow. After that i can start to see what the sensor outputs. The converter will then ready soon but the conversion of the locations will take a bit more time.

RealTadango, 2018-07-06:
It took a while, but the powerbox code works for a GPS module only now. The coordinates are correct to. It will simply relay the values it gets from a powerbox device. The GPS module must be configured for powerbox usage.

closed 2018-07-06 by RealTadango