RealTadango/FrSky #4: LView.lua still maintained ?

mipel85, 2019-05-06:
Hello,
I use this script (on my X10) for some time and find that we can not view the csv files of the previous year.
Is a modification possible to access all saved logs?
Thank you.
Mipel

RealTadango, 2019-05-06:
Yes, still maintained.

The issue is that there is no file listing option in Lua. The only way to check which files are present is to "guess" the filenames by trying the model name with a date format. There are 100 dates tried so you could see up to 100 days in the past in the selection. You can try to increase this number and see if the performance remains acceptable.

https://github.com/RealTadango/FrSky/blob/master/LView/LView.lua#L501 Increase the 100 to 1000 for example.

mipel85, 2019-05-06:
Thank you for responding quickly. I just changed L501 with "for i = 0, 600, 1 do ..." and I visualize well the 5 csv of 2018. If I try a value greater than 600, I have the message "CPU limit". So it will be like that. Do you plan to commit this change to other users?
Thanks again, and good development!
Mipel

RealTadango, 2019-05-06:
I will check how the 600 value works for me and will commit if there is little performance impact.

mipel85, 2019-05-06:
Of course ! It's ok on my X10 and Companion 2.2.3.
I start learning lua scripts and thank you for responding quickly to my request. Many more questions about lua, I love to learn!

mipel85, 2019-06-17:
Hello, 
No answer since 6 May, have you check my proposed modification ?

RealTadango, 2019-06-26:
Sorry, but i have added the modification now to 600 days. I wanted to add a loading screen for it since the startup is no 5 seconds.... maybe later.

mipel85, 2019-07-09:
Thank you, it's a good thing for all users.

ahmad-kassem, 2022-03-26:
Hello
the Lua is not working on my TX16S, it is always giving the error (unexpected symbol near '<')
view attached photo.

[image]

RealTadango, 2022-03-28:
> Hello the Lua is not working on my TX16S, it is always giving the error (unexpected symbol near '<') view attached photo.
> 
> [image]

Yes, this is still maintained. It works fine with OpenTX. What OS and Version are you using?

bionicbone, 2022-06-05:
Hi @RealTadango , can you confirm if LView works with EdgeTx, personally I have v2.7.1 installed ?
I did try, it intially showed me a corrupt screen and then goes on to state there are no logs for the current model.
Thanks, Kev.

RealTadango, 2022-06-08:
Hi @bionicbone, looks like it works fine with Edge TX. I just installed 2.7.1 on my RM TX16S and LView works as expected without errors. It does show a frame buffer when starting but that looks like an Edge TX bug. The no logs might be correct if there are no log files for the selected model in the last 600 days. It only looks for files for the current selected model.

bionicbone, 2022-06-09:
@RealTadango thanks for the support, I'll try again later. Thinking about it, it is possible the logs were probably pre-covid for the model I had selected.

ahmad-kassem, 2022-08-26:
@RealTadango 
hello i have updated my radio to EdgeTX and i noticed a problem in the LView Lua.
the problem is that now EdgeTX will not put a low dash between the words of the model name to replace spaces, for example i have a model called (TREX 700) normally in OpenTX when the log is created the model name is shown as TREX_700 in the created log file, but in EdgeTX the space is not replaced by a low dash when the log file is created and this prevents the LView Lua from seeing the log file.

can you please solve this problem because i really like this LUA.
Awesome work by the way, Thanks.

offer-shmuely, 2022-09-17:
Hi
I took the wonderful work of RealTadango!!!! and update the code to handle the edgeTx **_in my style_**
I am looking to testers, so if you like to test a second tool, I will be glad

https://github.com/offer-shmuely/edgetx-x10-scripts/blob/main/TOOLS/LogViewer.lua

and again, many thanks to RealTadango for the great code.

[image]