RealTadango/FrSky #5: Doesn't scroll - and only one value

PeterVRC, 2019-06-23:
Once the log/chart all loads I can;t find any way to load more than the one 'default' parameter it lists (out of 4 possible ones). eg it loads Amps and that is all.  The log file has a lot of other parameters in it.
Once the chart is displayed I cannot Scroll it.  I can change to Min/Max and alter those, but in Zoom it does not alter it, and in scroll it does not either.
When the log/chart was loading you could see a wide range of the data going past, but once it is finished you only see the last short section and cannot scroll to see any earlier time.
Horus X10S  OpenTX 2.2.3

RealTadango, 2019-06-26:
Can you attach the log file you loaded? Does this happen with all files / models?

PeterVRC, 2019-06-27:
The handful of models I have tried do the same.

But I found in this A-10 log that it does whatever it does at the graph screen, and the Zoom level follows the data size as it processes it. Eg by the end of the processing it is zoomed right in to a small section.

In the Hawk it did not do that.

Neither file can Scroll.

 

After file selection you can’t move off the Session Number field, and the Amps are in there,, but the other three fields are empty.

Pics included.

 

One pic shows a remaining “pink tip’ at the top left, where something (Amp data peak) drew above the chart, so it never gets cleared in future again.

 

 

Thanks for making the great program/script!1

 

Peterp

RealTadango, 2019-06-27:
Sorry, but i see no attachments or pic's when you reply from email.... :)

PeterVRC, 2019-06-27:
I attached them to this post:

 

 

https://www.rcgroups.com/forums/showpost.php?p=42181031 <https://www.rcgroups.com/forums/showpost.php?p=42181031&postcount=2> &postcount=2

RealTadango, 2019-07-08:
I will check the logs next week. The images look fine. The parts above and below the graph show the min and max positions of that value. 

With page up / down you can change the position in the menu to change the value.