RealTadango/FrSky #7: Jetcat telemetry sensor and rpm

alex998r, 2019-11-06:
I think it will be not much difficult to implement, a protol documentatio was provided early.
A rpm optical sensor would be nice also, frsky doesn't have it 
Thanks

RealTadango, 2019-11-06:
The difficult part about Jetcat is that i don't own a Jetcat turbine to test..... I someone can lend me an old one i can develop the code for it. Documentation is one thing, making it work another.

The optical sensor should not be hard to code (add a blade diver and count the pulses) but the hard part is getting a good reliable pulse from the optical sensor. That probably needs an analogue gain circuit. If you know a circuit online that works i will look into it.

alex998r, 2019-11-06:
I own 2 turbines, one with ecu V6 and another with ecu v10, so we can test both.
I sent you some time ago, the comlete documentation of the jetcat protocol ( request/response and payload ), maybe you missed it?
For the rpm sensor I wil do some researches now.
Many thanks for your reply

RealTadango, 2019-11-06:
I saw the documentation but without an ecu on my desk it is very hard to develop. I can loan a V10 from a friend so that should be an option. Later this year i will find some time to get the basics working (like rpm and egt) to make it compatible with the other ecu sensor.

alex998r, 2019-11-06:
Maybe this can be of help:
http://www.electronoobs.com/eng_arduino_tut15.php
The rpm sensor is not difficult, I think, to transpose to your library, maybe physical id 0x012 will be used along with a custom application id? Or the rpm sensor is already present in frsky code?

RealTadango, 2019-11-06:
I normally use 0x0500 for Rpm1 which a FrSky default id. Physical ID can be any. 

The provided example works with a manual gain which is the difficult part. Maybe a fixed gain can work.....