RealTadango/FrSky #9: Is Simple Sensor supported?

briannadeau, 2020-01-24:
Hi, I have tried the simple sensor example on arduino nano and don't see any output on the pin (watching on the o-scope).   Feels like I am missing something, but not sure what.   Looking at the code it's not clear hub.handle does anything in the "simple" case.    Any insights welcome.    thanks.

RealTadango, 2020-01-24:
That is correct :) The S.Port sensor is a slave and the receiver is a master device so the master need to poll the slave. 

hub.Handle reads the serial data and it there is a new request from the master a response will be send. With no master attached it doesn't do much.

briannadeau, 2020-01-25:
I have it working now, thanks for the help.   I had some issues with GPS and your library both wanting the serial support, the arduino library only wanted one active.   I swap back and forth now, which works fine.

RealTadango, 2020-01-25:
The sensor uses the software serial and you can use the hardware serial for the GPS.

closed 2020-01-26 by briannadeau