RealTadango/FrSky #10: Support for ESP32

realhuno, 2020-02-24:
Hi, first of all great work... i like ;-)

With my old Arduino Nano all working fine.
The esp32 only send one message and even when i reset... only one message

It would be nice when anybody can fix this, because i have plans with fpv racing and send laptimes to the transmitter and osd.

Thanks so much

RealTadango, 2020-02-25:
Ok, i will check  on an ESP8266 how it behaves. I don't have an ESP32 to test on. Can you share a simple sketch that i can use to verify or are the example sketches not working also?

realhuno, 2020-02-25:
Beautiful...  all your examples working fine with the arduino nano. But no chance with the esp. 
There are some ESP32 on the market with sketch upload issues.  Need to reset manual before the upload process begin.
If this happened.... the esp32 also send the s.port stream. But only when this happened.
[image]

RealTadango, 2020-03-01:
I tested using an ESP 12E (8266) at 160Mhz. I use D2 (gpio 4) and it works perfectly. Some GPIO ports do not work because the lack of interrupt support (D3 / D4 for example). 

You can try lowering the cpu speed to check if that works. If that makes a difference i might know a solution.

realhuno, 2020-03-07:
Hi, i can confirm..... working great on the esp8266.... but not with the esp32 ;-( no chance. I also test to slow down the cpu .... nothing working ;-(

RealTadango, 2020-03-07:
I have ordered an esp32 and will see what the difference is.

RealTadango, 2020-03-24:
Looks like there is no SoftwareSerial library for ESP32 by default. Which one did you use?

realhuno, 2020-03-24:
I use this one https://www.arduinolibraries.info/libraries/esp-software-serial

RealTadango, 2020-03-24:
So i found a SoftwareSerial library and got the behaviour replicated. Only one sensor frame send when the ESP is reset. Now i can start debugging :)

RealTadango, 2020-03-24:
And.... fixed :) I will check if this works fine with the Atmega's libraries also. Basically the ESP library needs to stop listening when i am transmitting. Somehow only switching the pin input mode causes the SoftwareSerial to crash. If all checks out i will push a new version tomorrow which should be available as a library update the next day.

RealTadango, 2020-03-25:
Version 1.0.2 has been created. It should be available when the Arduino server has scanned the new version

realhuno, 2020-04-26:
Now working. Thank you very much. good job

closed 2020-04-26 by realhuno