RealTadango/FrSky #14: GPS coordinates

mipel85, 2021-06-25:
Hello, 
I try to visualize a log containing GPS coordinates but the script stops and a message indicates "?: 0 attempt to compare number with nil"
I understand that there is an attempt to divide by zero on the coordinate values, is there a way to fix this?
I attach the log for example.
Thank you for the answer,
Michel
 
[ExtraMoteur-2021-06-25.csv](https://github.com/RealTadango/FrSky/files/6717733/ExtraMoteur-2021-06-25.csv)

RealTadango, 2021-06-25:
Hi Michel, In which fase did the script stop? Did you select the coordinates to display?

mipel85, 2021-06-25:
Thank you for this quick contact.
Yes, I took the rssi in first value then the GPS in second and the validation triggers this error.

mipel85, 2021-06-26:
additionally, the error is on line 784, probably a data conversion problem.

RealTadango, 2021-06-26:
The problem is that you cannot plot the GPS data. They are coordinates and not a single value that can be plotted.

mipel85, 2021-06-26:
I understand the principle, so there is no solution if not perhaps to prohibit the selection of this type of data as input values because it is the same thing for the dates.
I think this issue needs to be closed.
Thanks,

Michel