RealTadango/FrSky #15: Sensor lost alert

HirezRa, 2021-07-11:
I create an rpm sensor with Arduino NANO, and it work on
When I add a second sensor (pressure sensor BME280) to work together,
I received "sensor lost"

RPM + PRESSURE on the same Arduino

RealTadango, 2021-07-11:
Can you share your code? There can be many reasons....

HirezRa, 2021-07-11:
#include <SPort.h>                  //Include the SPort library
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_BME280.h>

int RcvSPin = 3;                // S.Port int pin 3
int Sns1Pin = 2;

SPortHub hub(0x12, RcvSPin);           //Hardware ID 0x12, Software serial pin 3
SimpleSPortSensor RPMsensor(0x0500);   //Sensor with ID 0x500
SimpleSPortSensor BROsensor(0x0100);   //Sensor with ID 0x100

#define SEALEVELPRESSURE_HPA (1013.25)
Adafruit_BME280 bme; // I2C

const unsigned long TmO = 100000;  
const byte numReadings = 2;
volatile unsigned long LTM;
volatile unsigned long PBP = TmO+1000;
volatile unsigned long PeriodAverage = TmO+1000;
unsigned long Feq_Tmp;
unsigned long RPM;
unsigned int PulseCounter = 1;
unsigned long PeriodSum;
unsigned long LastTimeCycleMeasure = LTM;  
unsigned long CurrentMicros = micros(); 
unsigned int AmountOfReadings = 1;
unsigned int ZeroDebouncingExtra; 
float filtered = 0;
float pressure = 0;

/****************************************************************/
void setup()  // Start of setup:
{  
  bme.begin(0x76, &Wire);
    
  hub.registerSensor(RPMsensor);       //Add sensor to the hub
  hub.registerSensor(BROsensor);       //Add sensor to the hub
  
  hub.begin();                      //Start listening
  attachInterrupt(digitalPinToInterrupt(Sns1Pin), Pulse_Event, RISING);  // Enable interruption pin 2 when going from LOW to HIGH.
}  // End of setup.
/****************************************************************/

void loop()  // Start of loop:
{
  bme.takeForcedMeasurement();
  
  LastTimeCycleMeasure = LTM;  // Store the LTM in a variable.
  CurrentMicros = micros();  
  if(CurrentMicros < LastTimeCycleMeasure)
  {
    LastTimeCycleMeasure = CurrentMicros;
  }
  Feq_Tmp = 10000000000 / PeriodAverage;  // Calculate the frequency using the period between pulses.
  if(PBP > TmO - ZeroDebouncingExtra || CurrentMicros - LastTimeCycleMeasure > TmO - ZeroDebouncingExtra)
  {  
    Feq_Tmp = 0;
    ZeroDebouncingExtra = 2000;
  }
  else
  {
    ZeroDebouncingExtra = 0;
  }
  
  // Calculate the RPM:
  RPM = Feq_Tmp / 60;
  
  RPMsensor.value = RPM;        /** S.PORT TELEMETRY **/
  
  pressure = (bme.readAltitude(SEALEVELPRESSURE_HPA));
 
  BROsensor.value = pressure;  /** S.PORT TELEMETRY **/
      
   hub.handle();                     
}  // End of loop.

 
void Pulse_Event()  // The interrupt runs this to calculate the period between pulses:
{
  PBP = micros() - LTM;  
  LTM = micros();  
  if(PulseCounter >= AmountOfReadings)  
  {
    PeriodAverage = PeriodSum / AmountOfReadings;  
    PulseCounter = 1;  
    PeriodSum = PBP;  
    int ReMapReads = map(PBP, 40000, 5000, 1, 10);  // Remap the period range to the reading range.
    ReMapReads = constrain(ReMapReads, 1, 10);  
    AmountOfReadings = ReMapReads;
  }
  else
  {
    PulseCounter++;
    PeriodSum = PeriodSum + PBP;  
  }
}  // End of Pulse_Event.