RealTadango/FrSky #19: Add explanation to readme about how to use the hardward serial.

RobinLeblond, 2021-11-29:
Hello, I didn't know how to ask this elsaway, but can you explan how to use it with the hardware serial instead of the sofware one ?
All your example look like useing the software serial, thanks.

RealTadango, 2021-11-29:
Better documentation is a good idea 😊

You can use hardware serial like this:

SPortHub hub(0x12, Serial);

RobinLeblond, 2021-11-29:
Wow, easy like that! Thanks

icebreaker-ch, 2023-12-28:
Unfortunately, this solution does not work due to several reasons:

1) The constructor of SPortHub with a Serial parameter is hidden in a strange conditional compilation statement:
```
 #ifdef Serial_
            SPortHub(int physicalId, Serial_& serial);
    #elif HardwareSerial
            SPortHub(int physicalId, HardwareSerial& serial);
    #endif
```
I think, the different constructors shall be all visible and overloaded.

2) Because of the conversion operator
`Serial_::operator bool()`
in the class
`Serial_`
the constructor used for
`SPortHub(0x12, Serial);`
is again;
`SPortHub(int physicalId, int softwarePin);`
which makes no sense at all, but we get no error message here.