RealTadango/FrSky #21: FrSky Ethos LUA grapher

mmasch, 2023-01-17:
Hi. 
Posted this question in RCGroups/Radios/Ethos LUA thread and a person there suggested you allready had 
developed a similar LUA for OpenTX. Any chance for a port to the Ethos OS?

Below is my query at RCGroups:
_Wondering if someone has considered creating a LUA to display logged content on the transmitter?
Ie. RSSI/VFR/Vario height et.al. from the logs from the current or another specified (selectable) flight?
I am not able to create LUA content, but thought I'd plant the idea.
Had visualized a full-screen where you could pick the log-file you wanted, and perhaps a dialogue box where
you could select one or several of the logged telemetry values and get it displayed graphically on the radio.
A little like the way OTX Companion does it.
Had been cool to have the ability to check for example the VFR and/og RSSI right after a flight._

Br. Morten Masch, Norway.

RealTadango, 2023-01-17:
Hi Morton,

I am working on it but ETHOS lacks some things i need. See this request: https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2243

When that part is implemented in ETHOS i will continue. I don't have a TX with ETHOS at the moment but that might change.

Kind regards, Herman

lthole, 2023-01-17:
Hi Herman

Perhaps a great first step would be for the user to browse to and select the log file to be viewed?

Kind regards,
Lothar

RealTadango, 2023-01-18:
> Hi Herman
> 
> Perhaps a great first step would be for the user to browse to and select the log file to be viewed?
> 
> Kind regards, Lothar

Sure, but how?

lthole, 2023-01-18:
I suppose I assumed that the standard io library might handle this.

I am no expert in Lua. I have written a simple status widget (it’s in the Ethos Lua thread) and I also helped Mike Shellim to port his adaptive trim script to a Horus widget.

RealTadango, 2023-01-18:
> I suppose I assumed that the standard io library might handle this. I am no expert in Lua. I have written a simple status widget (it’s in the Ethos Lua thread) and I also helped Mike Shellim to port his adaptive trim script to a Horus widget.
> […](#)

No, IO lib does not have this feature. Every example you see with file itteration relies on the operating system build in functions and the IO lib only has a passthrough option to interact with those. So nothing for ETHOS.

I don't consider myself a lua expert but i have found the limits very often. For my need there is no workaround at the moment. ETHOS needs to implement something for this.

lthole, 2023-01-18:
I will use what little influence I have with Bertrand to push for this.

lthole, 2023-02-14:
Hi Herman,This issue has been implemented, and is in the nightly release: https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2091RegardsLothar

RealTadango, 2023-02-14:
I don't see it in the change logs or lua documentation? Or is it in the lua samples somewhere?

lthole, 2023-02-14:
I found it in the assets for the nightly release here
https://github.com/FrSkyRC/ETHOS-Feedback-Community/releases/tag/nightly <https://github.com/FrSkyRC/ETHOS-Feedback-Community/releases/tag/nightly> 

Look in lua-doc.zip

To test it you would have to install the nightly release, which must be done manually for the time being, because Ethos Suite will only support nightlies from Suite 1.4.7.

lthole, 2023-02-14:
> Ah yes, function listFiles in class system. When there is a simulator that supports this i will check it. I don't have an ETHOS transmitter at the moment.

I will let you know. At the moment the simulator is at 1.4.4.

lthole, 2023-02-15:
FYI  https://www.rcgroups.com/forums/showpost.php?p=50548047&postcount=22710

RealTadango, 2023-02-16:
Ah, even better. They finally woke up :)