RealTadango/FrSky #22: PowerBox Telemetry and Servo Issues

MrOmbre, 2023-02-07:
It works and I get all Sensor Data from PowerBox Pioneer to my FrSky Horrus X10S Express, but if Converter is connected between PowerBox and Receiver, I get Servo Jitter on all 16 channels.
I had flash the powerboxsensor.ino to Arduino Pro Mini.

RealTadango, 2023-02-07:
How did you connect the ground wires? What RX input do you have, normal SBUS? Jitter should not be possible since the Powerbox receives the signal digitally. This sound like a bug in the powerbox firmware not able to handle telemetry timings correctly.

MrOmbre, 2023-02-07:
I connect ground from powerbox rx1 GND to arduino GND (serial input) and pin GND from arduino to GND receiver (telemetry input)
I have normal SBUS rx input

powerbox rx1 send SBUS and telemetry together on one wire
I must split with y-cable, cause FrSky receiver have separat input for SBUS and telemetry

RealTadango, 2023-02-07:
You can remove all arduino ground wires except for the sport. I power my arduino from SPort and have only the signal wire to the powerbox. Signal Ground is shared trough the powerbox.

MrOmbre, 2023-02-07:
nice to know, maybe its an ground issues (ground loop)

MrOmbre, 2023-02-07:
bad news, ground from powerbox side removed, but same issues

RealTadango, 2023-02-08:
Sounds like a powerbox firmware bug. If you remove telemetry it works fine? You could try another type of receiver to eliminate that....

MrOmbre, 2023-02-08:
I use an Archer R6 and tested with same result an Archer SR8
I think, there is nothing, I can do. Otherwise to by a new powerbox :-(